peterus/ESP-FTP-Server-Lib #2: New FTP-commands

akoro, 2021-02-13:
Hello!
I am very like your library. Thank you very much! It is very useful and flexible. It is good work
I suggest to include in the library some additional commands. It is OPTS, NOOP, FEAT and MLSD. I have made some code for it and add several lines to your file FTPConnection.cpp (See attached). I have already tested the corrected code for ESP32 and it does work.

[FTPConnection.zip](https://github.com/peterus/ESP-FTP-Server-Lib/files/5975849/FTPConnection.zip)
[MLSD.zip](https://github.com/peterus/ESP-FTP-Server-Lib/files/5975850/MLSD.zip)

peterus, 2021-02-19:
thank you very much for your changes, i will do a review this weekend and create a pull request.
if you want to have your name on the pull request i am also okay with waiting for your pull request 😉
if you need instructions how to create a pull request i can recommend to see some github videos on youtube.

akoro, 2021-02-27:
Pull request created

closed 2022-04-05 by peterus