peterus/ESP-FTP-Server-Lib #4: The library does not compile for ESP8266

akoro, 2021-02-27:
Unfortunately the library does not compile for ESP8266.
I suppose the difference is in filesystem classes for two platforms. Apparently, nothing can be done here in simple ways.

The compiler output is below

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.3) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP-FTP-Server-Lib>
|   |-- <ESP8266WiFi> 1.0
|-- <LittleFS(esp8266)> 0.1.0
Building in release mode
Compiling .pio\build\esp12e\src\main.cpp.o
Compiling .pio\build\esp12e\libc07\ESP-FTP-Server-Lib\ESP-FTP-Server-Lib.cpp.o
Compiling .pio\build\esp12e\libc07\ESP-FTP-Server-Lib\FTPConnection.cpp.o
Compiling .pio\build\esp12e\libc07\ESP-FTP-Server-Lib\FTPFilesystem.cpp.o
Archiving .pio\build\esp12e\liba31\libLittleFS.a
Compiling .pio\build\esp12e\FrameworkArduino\base64.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\cont.S.o
Compiling .pio\build\esp12e\FrameworkArduino\cont_util.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\core_esp8266_app_entry_noextra4k.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\core_esp8266_eboot_command.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\core_esp8266_features.cpp.o
Compiling .pio\build\esp12e\FrameworkArduino\core_esp8266_flash_quirks.cpp.o
In file included from lib\ESP-FTP-Server-Lib\src/FTPCommand.h:7:0,
                 from lib\ESP-FTP-Server-Lib\src/FTPConnection.h:7,
                 from lib\ESP-FTP-Server-Lib\src/ESP-FTP-Server-Lib.h:8,
                 from src\main.cpp:9:
lib\ESP-FTP-Server-Lib\src/FTPFilesystem.h:18:52: error: 'FILE_READ' was not declared in this scope
  File open(const String & path, const char* mode = FILE_READ);
                                                    ^
lib\ESP-FTP-Server-Lib\src/FTPFilesystem.h:50:10: error: 'boolean FTPFileImpl::isDirectory()' marked override, but does not override
  boolean isDirectory(void) override { return true; };
          ^
lib\ESP-FTP-Server-Lib\src/FTPFilesystem.h:51:18: error: 'fs::FileImplPtr FTPFileImpl::openNextFile(const char*)' marked override, but does not override
  fs::FileImplPtr openNextFile(const char* mode) override
                  ^
lib\ESP-FTP-Server-Lib\src/FTPFilesystem.h:61:7: error: 'void FTPFileImpl::rewindDirectory()' marked override, but does not override
  void rewindDirectory(void) override {};
       ^
lib\ESP-FTP-Server-Lib\src/FTPFilesystem.h:62:2: error: 'FTPFileImpl::operator bool()' marked override, but does not override
  operator bool() override { return false; };
  ^
lib\ESP-FTP-Server-Lib\src/FTPFilesystem.h: In member function 'fs::FileImplPtr FTPFileImpl::openNextFile(const char*)':
lib\ESP-FTP-Server-Lib\src/FTPFilesystem.h:59:46: error: cannot allocate an object of abstract type 'FTPFileImpl'
   return fs::FileImplPtr(new FTPFileImpl(next));
                                              ^
lib\ESP-FTP-Server-Lib\src/FTPFilesystem.h:36:7: note:   because the following virtual functions are pure within 'FTPFileImpl':
 class FTPFileImpl : public fs::FileImpl
       ^
In file included from C:\users\arkad\.platformio\packages\framework-arduinoespressif8266\libraries\LittleFS\src/LittleFS.h:32:0,
                 from src\main.cpp:8:
C:\users\arkad\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FSImpl.h:38:18: note:                                                                                                                          virtual bool fs::FileImpl::truncate(uint32_t)
     virtual bool truncate(uint32_t size) = 0;
                  ^
C:\users\arkad\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FSImpl.h:41:25: note:                                                                                                                          virtual const char* fs::FileImpl::fullName() const
[... 137 lines cut ...]
In file included from lib\ESP-FTP-Server-Lib\src\FTPConnection.cpp:10:0:
lib\ESP-FTP-Server-Lib\src\Commands/LIST.h: In member function 'virtual void LIST::run(FTPPath&, const std::vector<String>&)':
lib\ESP-FTP-Server-Lib\src\Commands/LIST.h:21:55: error: call to 'fs::File FTPFilesystem::open(const String&, const char*)' uses the default argument for parameter 2, which is not yet defined
   File dir = _Filesystem->open(WorkDirectory.getPath());
                                                       ^
C:\users\arkad\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FSImpl.h:41:25: note:                                                                                                                          virtual const char* fs::FileImpl::fullName() const
     virtual const char* fullName() const = 0;
                         ^
C:\users\arkad\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FSImpl.h:42:18: note:                                                                                                                          virtual bool fs::FileImpl::isFile() const
     virtual bool isFile() const = 0;
                  ^
C:\users\arkad\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FSImpl.h:43:18: note:                                                                                                                          virtual bool fs::FileImpl::isDirectory() const
     virtual bool isDirectory() const = 0;
                  ^
In file included from lib\ESP-FTP-Server-Lib\src\FTPConnection.cpp:14:0:
lib\ESP-FTP-Server-Lib\src\Commands/RETR.h: In member function 'virtual void RETR::run(FTPPath&, const std::vector<String>&)':
lib\ESP-FTP-Server-Lib\src\Commands/RETR.h:28:33: error: call to 'fs::File FTPFilesystem::open(const String&, const char*)' uses the default argument for parameter 2, which is not yet defined
   _file = _Filesystem->open(path);
                                 ^
*** [.pio\build\esp12e\libc07\ESP-FTP-Server-Lib\ESP-FTP-Server-Lib.cpp.o] Error 1
*** [.pio\build\esp12e\libc07\ESP-FTP-Server-Lib\FTPConnection.cpp.o] Error 1
================================================================================================== [FAILED] Took 2.53 seconds ==================================================================================================
The terminal process "C:\Users\arkad\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

akoro, 2021-03-02:
I introduced conditional translation, fixed all incompatibilities. Now everything compiles, but the server is not fully functional for ESP8266. When the command LIST is executed, a reboot occurs on the line of code **File f = dir.openNextFile();** in file LIST.h. So far I have not been able to figure out the reason for this.

peterus, 2021-03-06:
i will take a look on this when i have time, thanks for reporting this issue.

akoro, 2021-03-07:
OK. I will think about this issue too

maxgerhardt, 2021-05-03:
@akoro Where is that fixed version? At https://github.com/akoro/ESP-FTP-Server-Lib I still get all the same compiler errors :( 

```
Compiling .pio\build\huzzah\libe3f\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
In file included from C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SdFat\src/FatLib/FatFile.h:36:0,
                 from C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SdFat\src/FatLib/ArduinoFiles.h:33,
                 from C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SdFat\src/FatLib/FatLib.h:27,
                 from C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SdFat\src/SdFat.h:33,
                 from C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\SDFS\src/SDFS.h:36,
                 from C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\SD\src/SD.h:25,
                 from src\main.cpp:3:
C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SdFat\src/FatLib/FatApiConstants.h:104:19: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'const char*' [-fpermissive]
 #define O_READ    O_RDONLY
                   ^
Compiling .pio\build\huzzah\libe3f\ESP8266WiFi\WiFiUdp.cpp.o
C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\SD\src/SD.h:28:26: note: in expansion of macro 'O_READ'
 #define FILE_READ sdfat::O_READ
                          ^
.pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src/FTPFilesystem.h:18:52: note: in expansion of macro 'FILE_READ'
  File open(const String & path, const char* mode = FILE_READ);
                                                    ^
src\main.cpp: In function 'void setup()':
src\main.cpp:21:2: warning: 'SPIFFS' is deprecated (declared at C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
  SPIFFS.begin(true);
  ^
src\main.cpp:21:19: error: no matching function for call to 'fs::FS::begin(bool)'
  SPIFFS.begin(true);
                   ^
src\main.cpp:21:19: note: candidate is:
In file included from src\main.cpp:2:0:
C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:204:10: note: bool fs::FS::begin()
     bool begin();
          ^
C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:204:10: note:   candidate expects 0 arguments, 1 provided
src\main.cpp:22:21: error: no matching function for call to 'SPIClass::begin(int, int, int)'
Compiling .pio\build\huzzah\lib5fa\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\ESP-FTP-Server-Lib.cpp.o
  SPI.begin(14, 2, 15);
Compiling .pio\build\huzzah\lib5fa\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\FTPConnection.cpp.o
                     ^
src\main.cpp:22:21: note: candidate is:
In file included from C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\SDFS\src/SDFS.h:35:0,
Compiling .pio\build\huzzah\lib5fa\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\FTPFilesystem.cpp.o
                 from C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\SD\src/SD.h:25,
                 from src\main.cpp:3:
C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\SPI/SPI.h:56:8: note: void SPIClass::begin()
   void begin();
        ^
C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\libraries\SPI/SPI.h:56:8: note:   candidate expects 0 arguments, 3 provided
src\main.cpp:44:29: error: no matching function for call to 'FTPServer::addFilesystem(const char [3], SDClass*)'
  ftp.addFilesystem("SD", &SD);
                             ^
src\main.cpp:44:29: note: candidate is:
In file included from src\main.cpp:6:0:
.pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src/ESP-FTP-Server-Lib.h:20:7: note: void FTPServer::addFilesystem(String, fs::FS*)
  void addFilesystem(String Name, FS * const Filesystem);
       ^
.pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src/ESP-FTP-Server-Lib.h:20:7: note:   no known conversion for argument 2 from 'SDClass*' to 'fs::FS*'
src\main.cpp:45:31: warning: 'SPIFFS' is deprecated (declared at C:\Users\Max\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
  ftp.addFilesystem("SPIFFS", &SPIFFS);
                               ^
*** [.pio\build\huzzah\src\main.cpp.o] Error 1
In file included from .pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src\FTPCommand.h:7:0,
                 from .pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src\FTPConnection.h:7,
                 from .pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src\ESP-FTP-Server-Lib.h:8,
                 from .pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src\ESP-FTP-Server-Lib.cpp:1:
.pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src\FTPFilesystem.h:18:52: error: 'FILE_READ' was not declared in this scope
  File open(const String & path, const char* mode = FILE_READ);
[... cut ...]
.pio\libdeps\huzzah\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\src\Commands/RETR.h:28:33: error: call to 'fs::File FTPFilesystem::open(const String&, const char*)' uses the default argument for parameter 2, which is not yet defined
   _file = _Filesystem->open(path);
                                 ^
*** [.pio\build\huzzah\lib5fa\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\FTPFilesystem.cpp.o] Error 1
*** [.pio\build\huzzah\lib5fa\ESP-FTP-Server-Lib@src-c294c6a1a1101ba448c99eb922af932a\FTPConnection.cpp.o] Error 1
```

peterus, 2022-04-05:
finally fixed in release 0.9.9:  https://github.com/peterus/ESP-FTP-Server-Lib/releases/tag/v0.9.9

closed 2022-04-05 by peterus