peterus/ESP-FTP-Server-Lib #12: compilation error: pure virtual function "fs::FileImpl::path" has no overrider

lszlfrng, 2022-05-05:
Hello,
compiling the example (test.cpp) I got error:

```
In file included from lib/ESP-FTP-Server-Lib/src/FTPCommand.h:7,
                 from lib/ESP-FTP-Server-Lib/src/FTPConnection.h:7,
                 from lib/ESP-FTP-Server-Lib/src/ESP-FTP-Server-Lib.h:8,
                 from src/main.cpp:10:
lib/ESP-FTP-Server-Lib/src/FTPFilesystem.h: In member function 'virtual fs::FileImplPtr FTPFileImpl::openNextFile(const char*)':
lib/ESP-FTP-Server-Lib/src/FTPFilesystem.h:79:46: error: invalid new-expression of abstract class type 'FTPFileImpl'
   return fs::FileImplPtr(new FTPFileImpl(next));
                                              ^
lib/ESP-FTP-Server-Lib/src/FTPFilesystem.h:40:7: note:   because the following virtual functions are pure within 'FTPFileImpl':
 class FTPFileImpl : public fs::FileImpl
       ^~~~~~~~~~~
In file included from lib/ESP-FTP-Server-Lib/src/FTPFilesystem.h:6,
                 from lib/ESP-FTP-Server-Lib/src/FTPCommand.h:7,
                 from lib/ESP-FTP-Server-Lib/src/FTPConnection.h:7,
                 from lib/ESP-FTP-Server-Lib/src/ESP-FTP-Server-Lib.h:8,
                 from src/main.cpp:10:
/home/lfirnaga/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FSImpl.h:41:25: note: 	'virtual const char* fs::FileImpl::path() const'
     virtual const char* path() const = 0;
                         ^~~~
*** [.pio/build/esp32dev/src/main.cpp.o] Error 1
```

Could you please help what is the solution?

The environment: 
> platformio
> platform = espressif32
> board = esp32dev
> framework = arduino
> ide: eclipse

slaesh, 2022-05-07:
@lszlfrng take a look here: https://github.com/peterus/ESP-FTP-Server-Lib/pull/13/files

you can just fix it on your site if you copy that new line into the `FTPFilesystem.h` in your `libdeps/--platform--/ESP-FTP-Server-Lib/` folder

lszlfrng, 2022-05-09:
After the modification, the source code has been compiled without error.