peterus/ESP-FTP-Server-Lib #17: download problem

josefgull, 2022-07-15:
hi i am on an mac an use filzilla and the examplesketch on esp32 without sd
- it connects and i see my stored file inside spiffs folder
- when i try to download an empty file is downloaded (it shows in filezilla that the file shouldnt be empty)
- filzilla ask me to overwirte the empty file and then its still empty

what am i missing ?  is there a permission in the esp32 code i can activate?
or are the filzilla setttings wrong?

i kindly appreciate your help
[image]

[image]

josefgull, 2022-07-15:
still not working :/ i started everything new and changed to active mode :/

peterus, 2022-07-15:
please check again the serial console.

josefgull, 2022-07-16:
sorry

```
Connected to Ananasbude
IP address: 192.168.8.50
...---'''---...---'''---...---'''---...
New Connection from 192.168.8.54:61205
USER ftp
PASS ftp
SYST
FEAT
CWD /SPIFFS
PWD
TYPE I
PORT 192,168,8,54,239,22
MLSD
TYPE A
PORT 192,168,8,54,239,23
RETR vibrations.txt
PORT 192,168,8,54,239,25
RETR vibrations.txt
PORT 192,168,8,54,239,26
RETR vibrations.txt
PORT 192,168,8,54,239,27
RETR vibrations.txt
```