peterus/ESP-FTP-Server-Lib #19: Question FTP Status

DB8BH, 2022-11-12:
Version 0.11.0
ESP32
PlatformIO  (most recent version)

Hi,
I’m using the ftp-server with  2 file systems in parallel, LittleFS and SD.
Everything works as expected. Thanks for your great contribution.

Question:
Is there a way to retrieve some status information about the ftp transfer from inside the ESP program?  For instance the program needs to react on "New connection" resp. "Connection close". But info about transfer completion would also be useful.

Do you have any hint?

Thanks in advance

peterus, 2022-11-16:
the only function which will give a little bit of more information is the function `countConnections()`

In one of my projects I am using it like this: https://github.com/lora-aprs/LoRa_APRS_iGate/blob/e4cfd53f8aed83a8db156264f8ef803c58c7bb2f/src/TaskFTP.cpp#L32

but you are right, some more informations to get would be nice.

DB8BH, 2022-11-17:
Thanks a lot for your response.
I just tried your proposal. In principal it does what I'm looking for. Within the application I can determine if a client is connected to the server or not.  So, Question answered.
Anyway, may be there is a chance for some enhancement in the future. May be a callback that shows the essential steps of the data transfer.

Thanks again for your contribution so far.