peterus/ESP-FTP-Server-Lib #23: Error compiling for ESP32 Arduino IDE

mydroidbox, 2023-01-06:
Appreciate the library trying to get it working but receiving compile errors below:

In file included from C:\Users\val\OneDrive\Documents\Arduino\libraries\ESP-FTP-Server-Lib-master\src/FTPCommand.h:7:0,
                 from C:\Users\val\OneDrive\Documents\Arduino\libraries\ESP-FTP-Server-Lib-master\src/FTPConnection.h:7,
                 from C:\Users\val\OneDrive\Documents\Arduino\libraries\ESP-FTP-Server-Lib-master\src/ESP-FTP-Server-Lib.h:8,
                 from C:\Users\val\OneDrive\Documents\Arduino\YEA\YEA.ino:10:
C:\Users\val\OneDrive\Documents\Arduino\libraries\ESP-FTP-Server-Lib-master\src/FTPFilesystem.h:61:52: error: missing binary operator before token "("
  #if ESP_ARDUINO_VERSION >= ESP_ARDUINO_VERSION_VAL(2, 0, 3)

mydroidbox, 2023-01-16:
Btw I got your 0.9.9 version to compile on arduino IDE. Runs beautifully.
I've tried  to convert it to a soft AP but no luck yet. I can log in but can't access files.
I added: 
    WiFi.mode(WIFI_AP);
    WiFi.softAP(ssid, password);
    delay(1000);
    IPAddress myIP = WiFi.softAPIP();
    Serial.print("AP IP address: ");
    Serial.println(myIP);
return from filezilla:
Command: PORT 192,168,4,2,205,59
Response: 200 PORT command successful
Command: MLSD
Response: 425 No data connection
Error: Failed to retrieve directory listing

Hatuzilla, 2023-02-16:
Hi, I can confirm I get the same error as post #1

closed 2023-06-09 by peterus