peterus/ESP-FTP-Server-Lib #41: Spaces and Special Characters in FIlenames

Joe91, 2023-07-08:
First of all: Thanks a lot! It works really super stable and is very fast in the upload compared to other solutions.

The issue I have:
If the files contain spaces (or other non-standard characeters) the file won't be created on the filesystem with this name. 
With spaces it only takes the first part to the first space.
With special characters (like "ö" or "ß" or so) it does not work at all.
In the other direction (Download from FTP server) the same thing.

Is there a fix or workaround already available?

Joe91, 2023-07-08:
Update:
the reason for the spaces is https://github.com/peterus/ESP-FTP-Server-Lib/blob/master/src/FTPConnection.cpp#L74
when I replace this line with this, the spaces in folders and filenames work again:
`_LineSplited = {_Line.substring(0, index_separator), _Line.substring(index_separator + 1, _Line.length())};`

peterus, 2023-07-08:
@Joe91 thank you for taking a look into it!
can you create a merge request? than I will merge your change 😉

Joe91, 2023-07-08:
Yeah, of course.
I want to check some things first. 
Also got UTF-8 support working but I want to check if there are some commands, that might need some special handling.
Will create a PR this evening if everything works fine :-)

peterus, 2023-07-08:
@Joe91 thanks for your PR, it is already merged ;) new version will come soon!

closed 2023-07-08 by peterus