peterus/ESP-FTP-Server-Lib #44: File transfer successful, but file size less than the original

yoppy-tjhin, 2023-07-31:
Hi,

Thanks for your wonderful FTP server.
I was testing with ESP32 module (38 pin, Chip is ESP32-D0WDQ6 (revision v1.0)), ESP Arduino 2.0.10, and the latest ESP-FTP-Server-Lib.
Either using WinSCP or Filezilla, files can be uploaded or downloaded successfully without error notifications.

However, for unknown reasons some files are actually not completely transferred, as indicated by the size is less than the original file.
This is an example file: 
https://drive.google.com/file/d/1bdaU4GCI2FMhASfk7SkDdKOKvaTC2E6O/view?usp=sharing

The original size is 573,825 bytes. But when it is uploaded to the SD Card of ESP32 FTP server, it becomes 573.440 bytes.
So the file becomes defective, and cannot be opened properly.

What may I have done wrong? Thanks for your guidance.

Gabry1973, 2023-08-11:
Hi,

I think the issue lies in line 68 of the FTPCommand.h file, in function data_read. The connection is being interrupted before the data buffer can be emptied.
I have modified the function with the following code:

```c++
int data_read(uint8_t *c, size_t l) {
    if ((_DataConnection != 0) && (_DataConnection->available()>0))
        return _DataConnection->readBytes(c, l);
    return 0;
}
``` 

I hope I have been helpful.

DB8BH, 2023-09-14:
Hi,

I encountered the same problem as yoppy-tjhin.
Some binary files are transferred completely and successfully according to the FTP client, but were stored incompletely on the SD card. 
The problem seems to be independent of the absolute size of the files. Also the version of the Arduino framework does not matter. I was able to observe the problem with the Arduino versions 6.3.0, 6.3.2 and 6.4.0 as well.

I use the FTP server to transfer ESP32 firmware updates to the SD card of the target system.
An example of a firmware file that is not saved correctly can be found here:
_https://www.dropbox.com/scl/fi/aal4v8jmayfujp6sts6xi/original_nok.bin?rlkey=0gg3ixbvwppu9lks0yhgtxp11&dl=0_

The development system (PlatformIO 6.1.11, Arduino 6.3.2) creates this file as firmware.bin with a size of  872.848 Byte. 
But only 872.448 Byte are stored on the SD card.

If I stuff the original file at the end with 0xff to a size of at least 873,096 Bytes, it is saved correct and can be used as a firmware file. It can be flashed into the ESP32 as a firmware update.
_https://www.dropbox.com/scl/fi/pyx1f54al1bqh1g8cze9m/stuffed_ok.bin?rlkey=v5a5liloc300c1ba8ys22v0y0&dl=0_

I hope these observations may help with error analysis and correction.

Gabry1973, 2023-09-14:
Hi,

From the analysis I conducted, I noticed that the file is being truncated because the client closes the connection to the FTP server before it finishes writing the entire buffer to disk.
Programs like FileZilla, when transferring files, open a second connection and close it as soon as they finish transferring the data.
To solve the problem, it's necessary to complete writing the received data even if the connection is closed prematurely.
This can be achieved in the data_read function in FTPCommand.h. If the client disconnects or loses the connection the function currently returns 0.
In reality, it should continue to execute DataConnection->readBytes until it returns 0.

I hope I have been helpful.

peterus, 2023-09-14:
I will take a look on it when I have time.

If somebody else has a fixed and the fix is confirmed by someone else i will merge the provided fix.

peterus, 2023-09-14:
looks like the calculation of the file size on the server is wrong. at least my file size is not correctly shown in filezilla.

try this: upload the file - than download it again to your pc. check now the file size on your pc and make a diff between this two files.

peterus, 2023-09-14:
fix: https://github.com/peterus/ESP-FTP-Server-Lib/pull/48

DB8BH, 2023-09-14:
Thanks for your quick response.

Usually I use the TotalCommander FTP-Client, but I now doublechecked with Filezilla Client.
Same result, unfortunately.

I uploaded the original file with 872.848 Byte. Server showed 872.448 Byte.
Download this file, still 872.448 Byte. Obviously it's truncate.
CRC32 comparation of the file before upload and after download confirmed that issue.

DB8BH, 2023-09-15:
Hi,

I've modified the "FTPCommand.h" according und fix #48.

With that fix installed I can upload the original file (872.848 Byte) with FileZilla and it is stored correctly. Download and CRC32 check confirmed that it is ok. The fix seems to be effective so far. Thanks. But it also seems to have a side effect.

I usually use the FTP-Client integrated in TotalCommander. Before Fix #48 the behaviour was identical with FileZilla. After Fix #48 the upload with the TC FTP-Client is corrupted. The file is stored with 0 Byte length and the subsequent download is also 0 Byte.
No idea, if this is still part of the same problem.

Thanks again