peterus/ESP-FTP-Server-Lib #45: FTP Server crashes on "ls"-command

uhu1333, 2023-08-29:
I compiled your example on ESP8266 with Platform IO as is. SPIFFS was built and uploaded. My FTP client connects successfully. But on a "ls"-command the server crashes with reboot in file NLST.h around line 27 in statement

File f = dir.openNextFile();

with
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40205725 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

Any idea?

peterus, 2023-08-29:
@uhu1333 please post a full example, the log of the ftp client and when possible the log of the ftp server (just if there is a log output regarding ftp).
Its a long time since i checked on ESP8266.

uhu1333, 2023-08-29:
Thank you

Client log:
 ftp 192.168.1.119
Verbindung mit 192.168.1.119 wurde hergestellt.
220--- Welcome to FTP Server for ESP32 ---
220---        By Peter Buchegger       ---
220 --           Version 0.1           ---
200 Ok
Benutzer (192.168.1.119:(none)): ftp
331 OK. Password required
Kennwort:
230 OK.
ftp> ls
200 PORT command successful

Server log:
Connected to xxxx
IP address: 192.168.1.119
...---'''---...---'''---...---'''---...
New Connection from 192.168.1.191:50406
OPTS UTF8 ON
USER ftp
PASS ftp
PORT 192,168,1,191,196,231
NLST

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40205725 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[main.txt](https://github.com/peterus/ESP-FTP-Server-Lib/files/12464396/main.txt)

And the Soourc as .txt (instead of .cpp)